yaqwsx/PcbDraw #157: Breaking change in lxml 5.0.0

M0WUT, 2024-01-01:
Found as part of using kikit. Running `pcbdraw plot --vcuts=Cmts.User --side=front <path-to-pcb> <path-to-png-out>` completes successfully with lxml 4.9.3 installed. Upgrading lxml to 5.0.0 (which was released on 29th December2023) gives the following:

Traceback (most recent call last):
  File "/home/runner/.local/bin/pcbdraw", line 8, in <module>
    sys.exit(run())
  File "/usr/lib/python3/dist-packages/click/core.py", line 1128, in __call__
    return self.main(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/click/core.py", line 1053, in main
    rv = self.invoke(ctx)
  File "/usr/lib/python3/dist-packages/click/core.py", line 1659, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/lib/python3/dist-packages/click/core.py", line 1395, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/lib/python3/dist-packages/click/core.py", line 754, in invoke
    return __callback(*args, **kwargs)
  File "/home/runner/.local/lib/python3.10/site-packages/pcbdraw/ui.py", line 180, in plot
    image = plotter.plot()
  File "/home/runner/.local/lib/python3.10/site-packages/pcbdraw/plot.py", line 1052, in plot
    self._setup_document(self.render_back, self.mirror)
  File "/home/runner/.local/lib/python3.10/site-packages/pcbdraw/plot.py", line 1311, in _setup_document
    self._document = empty_svg(
  File "/home/runner/.local/lib/python3.10/site-packages/pcbdraw/plot.py", line 394, in empty_svg
    document = etree.ElementTree(etree.fromstring(
  File "src/lxml/etree.pyx", line 3265, in lxml.etree.fromstring
  File "src/lxml/parser.pxi", line 1975, in lxml.etree._parseMemoryDocument
  File "src/lxml/parser.pxi", line 1855, in lxml.etree._parseDoc
  File "src/lxml/parser.pxi", line 1086, in lxml.etree._BaseParser._parseUnicodeDoc
  File "src/lxml/parser.pxi", line 618, in lxml.etree._ParserContext._handleParseResultDoc
  File "src/lxml/parser.pxi", line 728, in lxml.etree._handleParseResult
  File "src/lxml/parser.pxi", line 657, in lxml.etree._raiseParseError
  File "<string>", line 1
lxml.etree.XMLSyntaxError: Blank needed here, line 1, column 21

Uninstalling lxml and force installing version 4.9.3 causes it to work again.

Thanks,
Dan

Electro707, 2024-01-02:
Can confirm this is an issue.

Trying to investigate for a bit, but I am not familiar with the lxml library

yaqwsx, 2024-01-02:
We are not alone: https://github.com/python-openxml/python-docx/issues/1302

LXML refuses to parse our empty SVG template: https://github.com/yaqwsx/PcbDraw/blob/0a137fe1d82eefb7e8c1ceacd642d14f13fcde44/pcbdraw/plot.py#L396-L404

It complains that a space is missing before `standalone`... which there is. I haven't been able to identify the root cause at the moment.

ofaurax, 2024-01-15:
It should work if you remove the `standalone="no"`

ofaurax, 2024-01-15:
I reported it here: https://gitlab.gnome.org/GNOME/libxml2/-/issues/661