yaqwsx/PcbDraw #160: Failed installation - Kicad 7.0,, Windows machine

Yoliq, 2024-01-28:
Hi all, 

currently I am having issues installing Pcbdraw for Kicad 7.0 on windows PC. 

Below you can find the log from installation. At the end there are some notes from my side.

```
************************************
* KiCad 7.0 Command Prompt
************************************
You may now invoke python or pip targeting kicad's install

C:\Users\danie\Documents\KiCad\7.0>pip install PcbDraw
Defaulting to user installation because normal site-packages is not writeable
Collecting PcbDraw
  Downloading PcbDraw-1.1.2-py3-none-any.whl (1.1 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 1.1/1.1 MB 4.9 MB/s eta 0:00:00
Collecting Pillow~=9.0 (from PcbDraw)
  Downloading Pillow-9.5.0-cp39-cp39-win_amd64.whl (2.5 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 2.5/2.5 MB 40.8 MB/s eta 0:00:00
Requirement already satisfied: click>=7.1 in c:\users\danie\documents\kicad\7.0\3rdparty\python39\site-packages (from PcbDraw) (8.1.7)
Collecting lxml (from PcbDraw)
  Obtaining dependency information for lxml from https://files.pythonhosted.org/packages/cd/54/9c03f5c83dec8d611f33f5b0e796eed718cf062b2cf5171188b923ae48ad/lxml-5.1.0-cp39-cp39-win_amd64.whl.metadata
  Downloading lxml-5.1.0-cp39-cp39-win_amd64.whl.metadata (3.6 kB)
Collecting mistune>=2.0.2 (from PcbDraw)
  Obtaining dependency information for mistune>=2.0.2 from https://files.pythonhosted.org/packages/f0/74/c95adcdf032956d9ef6c89a9b8a5152bf73915f8c633f3e3d88d06bd699c/mistune-3.0.2-py3-none-any.whl.metadata
  Downloading mistune-3.0.2-py3-none-any.whl.metadata (1.7 kB)
Requirement already satisfied: numpy in c:\program files\kicad\7.0\bin\lib\site-packages (from PcbDraw) (1.26.0)
Requirement already satisfied: pcbnewTransition<=0.4,>=0.3 in c:\users\danie\documents\kicad\7.0\3rdparty\python39\site-packages (from PcbDraw) (0.3.4)
Requirement already satisfied: pybars3 in c:\users\danie\documents\kicad\7.0\3rdparty\python39\site-packages (from PcbDraw) (0.9.7)
Collecting pyyaml (from PcbDraw)
  Obtaining dependency information for pyyaml from https://files.pythonhosted.org/packages/84/4d/82704d1ab9290b03da94e6425f5e87396b999fd7eb8e08f3a92c158402bf/PyYAML-6.0.1-cp39-cp39-win_amd64.whl.metadata
  Downloading PyYAML-6.0.1-cp39-cp39-win_amd64.whl.metadata (2.1 kB)
Collecting svgpathtools==1.4.1 (from PcbDraw)
  Downloading svgpathtools-1.4.1-py2.py3-none-any.whl (63 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 63.1/63.1 kB 3.5 MB/s eta 0:00:00
Collecting LnkParse3 (from PcbDraw)
  Obtaining dependency information for LnkParse3 from https://files.pythonhosted.org/packages/48/49/bc9376b66b92ed21623f278ba1428091f8f9a0a0709069e8689669e056bd/LnkParse3-1.3.3-py3-none-any.whl.metadata
  Downloading LnkParse3-1.3.3-py3-none-any.whl.metadata (8.0 kB)
Collecting svgwrite (from svgpathtools==1.4.1->PcbDraw)
  Downloading svgwrite-1.4.3-py3-none-any.whl (67 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 67.1/67.1 kB ? eta 0:00:00
Requirement already satisfied: colorama in c:\users\danie\documents\kicad\7.0\3rdparty\python39\site-packages (from click>=7.1->PcbDraw) (0.4.6)
Requirement already satisfied: PyMeta3>=0.5.1 in c:\users\danie\documents\kicad\7.0\3rdparty\python39\site-packages (from pybars3->PcbDraw) (0.5.1)
Downloading mistune-3.0.2-py3-none-any.whl (47 kB)
   ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 48.0/48.0 kB 2.5 MB/s eta 0:00:00
Downloading LnkParse3-1.3.3-py3-none-any.whl (47 kB)
   ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 47.4/47.4 kB 2.3 MB/s eta 0:00:00
Downloading lxml-5.1.0-cp39-cp39-win_amd64.whl (3.9 MB)
   ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 3.9/3.9 MB 41.4 MB/s eta 0:00:00
Downloading PyYAML-6.0.1-cp39-cp39-win_amd64.whl (152 kB)
   ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 152.8/152.8 kB 9.5 MB/s eta 0:00:00
Installing collected packages: LnkParse3, svgwrite, pyyaml, Pillow, mistune, lxml, svgpathtools, PcbDraw
Successfully installed LnkParse3-1.3.3 PcbDraw-1.1.2 Pillow-9.5.0 lxml-5.1.0 mistune-3.0.2 pyyaml-6.0.1 svgpathtools-1.4.1 svgwrite-1.4.3

[notice] A new release of pip is available: 23.2.1 -> 23.3.2
[notice] To update, run: python.exe -m pip install --upgrade pip

C:\Users\danie\Documents\KiCad\7.0>pcbdraw --help
Traceback (most recent call last):
  File "C:\Program Files\KiCad\7.0\bin\Lib\runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Program Files\KiCad\7.0\bin\Lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "C:\Users\danie\Documents\KiCad\7.0\3rdparty\Python39\Scripts\pcbdraw.exe\__main__.py", line 4, in <module>
  File "C:\Users\danie\Documents\KiCad\7.0\3rdparty\Python39\site-packages\pcbdraw\ui.py", line 15, in <module>
    from .populate import populate
  File "C:\Users\danie\Documents\KiCad\7.0\3rdparty\Python39\site-packages\pcbdraw\populate.py", line 16, in <module>
    from mistune.plugins.table import plugin_table # type: ignore
ImportError: cannot import name 'plugin_table' from 'mistune.plugins.table' (C:\Users\danie\Documents\KiCad\7.0\3rdparty\Python39\site-packages\mistune\plugins\table.py)
[... 153 lines cut ...]
        Content Cell  | Content Cell

    :param md: Markdown instance
    """
    md.block.register('table', TABLE_PATTERN, parse_table, before='paragraph')
    md.block.register('nptable', NP_TABLE_PATTERN, parse_nptable, before='paragraph')

    if md.renderer and md.renderer.NAME == 'html':
        md.renderer.register('table', render_table)
        md.renderer.register('table_head', render_table_head)
        md.renderer.register('table_body', render_table_body)
        md.renderer.register('table_row', render_table_row)
        md.renderer.register('table_cell', render_table_cell)

def table_in_quote(md):
    """Enable table plugin in block quotes."""
    md.block.insert_rule(md.block.block_quote_rules, 'table', before='paragraph')
    md.block.insert_rule(md.block.block_quote_rules, 'nptable', before='paragraph')

def table_in_list(md):
    """Enable table plugin in list."""
    md.block.insert_rule(md.block.list_rules, 'table', before='paragraph')
    md.block.insert_rule(md.block.list_rules, 'nptable', before='paragraph')

```

yaqwsx, 2024-01-29:
The problem you observe seems to be incorrect version of the library `mistune` installed on your system.

Yoliq, 2024-01-29:
Could you please provide me with `mistune` version that you use? I had version _3.0.2_. Tried to uninstall and install different versions but same result with _3.0.1_, _3.0.0_.

Also checked [github history](https://github.com/lepture/mistune) of the table.py file in hopes of finding the right version including plugin_table object but no luck.

yaqwsx, 2024-01-30:
The installation process clearly requires version 2 (see https://github.com/yaqwsx/PcbDraw/blob/0a137fe1d82eefb7e8c1ceacd642d14f13fcde44/setup.py#L29). So I am wondering how did pip even manage to install such a dependency mismatch.

Yoliq, 2024-01-31:
Thank you, issue fixed and pcbdraw works with 2.0.5

I also wonder what did pip do :)

I am closing the ticket.

```
C:\Users\danie\Documents\KiCad\7.0>pip install mistune==2.0.5
Defaulting to user installation because normal site-packages is not writeable
Collecting mistune==2.0.5
  Downloading mistune-2.0.5-py2.py3-none-any.whl (24 kB)
Installing collected packages: mistune
ERROR: pip's dependency resolver does not currently take into account all the packages that are installed. This behaviour is the source of the following dependency conflicts.
pcbdraw 1.1.2 requires Pillow~=9.0, but you have pillow 10.0.1 which is incompatible.
Successfully installed mistune-2.0.5

[notice] A new release of pip is available: 23.2.1 -> 23.3.2
[notice] To update, run: python.exe -m pip install --upgrade pip

C:\Users\danie\Documents\KiCad\7.0>mistune --version
'mistune' is not recognized as an internal or external command,
operable program or batch file.

C:\Users\danie\Documents\KiCad\7.0>pip install mistune
Defaulting to user installation because normal site-packages is not writeable
Requirement already satisfied: mistune in c:\users\danie\documents\kicad\7.0\3rdparty\python39\site-packages (2.0.5)

[notice] A new release of pip is available: 23.2.1 -> 23.3.2
[notice] To update, run: python.exe -m pip install --upgrade pip

C:\Users\danie\Documents\KiCad\7.0>pcbdraw --help
Usage: pcbdraw [OPTIONS] COMMAND [ARGS]...

  PcbDraw generates images of KiCAD PCBs

Options:
  --version  Show the version and exit.
  --help     Show this message and exit.

Commands:
  plot      Create a stylized drawing of the PCB.
  populate  Create assembly step-by-step guides
  render    Create a rendered image of the PCB using KiCAD's 3D Viewer

C:\Users\danie\Documents\KiCad\7.0>pcbdraw --version
pcbdraw, version 1.1.2

C:\Users\danie\Documents\KiCad\7.0>
```

closed 2024-01-31 by Yoliq