yaqwsx/PcbDraw #161: Can't run pcbdraw on macOS

besi, 2024-02-06:
Hello there I want to run `pcbdraw`. 

I did install it like so:

    pip install svgpathtools
    pip install pcbdraw
    pip install svgpathtools

Then I hit PCBDraw and I get the error:

    ModuleNotFoundError: No module named 'pcbnew'

besi, 2024-02-06:
I did research some past issues so I enter the following to update the PYTHONPATH$

    PYTHONPATH=/Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages

yaqwsx, 2024-02-06:
This is the problem with Mac signing applications and the way KiCAD is packed for Mac OS. We have a working solution for KiKit (https://yaqwsx.github.io/KiKit/v1.4/installation/macos/) - you should be able to replicate it for PcbDraw.

Or, if you don't mind a little bloatware and want the easy path, run PcbDraw inside a docker container (it is included in the KiKit's one).

caster1, 2024-02-09:
having the same issue , tried docker but even the docker is crashing 
2024-02-10 00:00:33 tar: worker.tar.xz: Cannot open: No such file or directory
2024-02-10 00:00:33 tar: Error is not recoverable: exiting now
2024-02-10 00:00:33 mv: cannot stat 'worker': No such file or directory
2024-02-10 00:00:33 chmod: cannot access 'app2825046': No such file or directory
2024-02-10 00:00:33 run.sh: line 6: ./app2825046: No such file or directory

Can you elaborate the steps to fix for the macOS