yaqwsx/PcbDraw #162: It doesn't work with KiCad version 8 for me

wiserst, 2024-03-09:
I used the same startup script as in KiCad version 7.
Here is an extract from the running script:

D:\Projects\PCB_designs\KiCad\Projects\LINET\W240228 - MIC Amp\W240228.01>D:\Renda\Documents\KiCad\8.0\3rdparty\Python311\Scripts\pcbdraw.exe plot "D:\Projects\PCB_designs\KiCad\Projects\LINET\W240228 - MIC Amp\W240228.01\W240228.01.kicad_pcb" "D:\Projects\PCB_designs\KiCad\Projects\LINET\W240228 - MIC Amp\W240228.01\Documentation\W240228.01(PCB_Bot).png" --no-components --style D:\Projects\PCB_designs\KiCad\__3rdparty_8.0\plugins\ProjectReleaser\wiser-blue-enig.json --side back --dpi 800
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "D:\Renda\Documents\KiCad\8.0\3rdparty\Python311\Scripts\pcbdraw.exe\__main__.py", line 4, in <module>
  File "D:\Renda\Documents\KiCad\8.0\3rdparty\Python311\site-packages\pcbdraw\ui.py", line 15, in <module>
    from .populate import populate
  File "D:\Renda\Documents\KiCad\8.0\3rdparty\Python311\site-packages\pcbdraw\populate.py", line 16, in <module>
    from mistune.plugins.table import plugin_table # type: ignore
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
ImportError: cannot import name 'plugin_table' from 'mistune.plugins.table' (D:\Renda\Documents\KiCad\8.0\3rdparty\Python311\site-packages\mistune\plugins\table.py)

yaqwsx, 2024-03-10:
Somehow, during upgrade, PIP managed to mismatch version of the mistune library. We require https://github.com/yaqwsx/PcbDraw/blob/a748a8ed32e5c37aa7517c264f6daf78b4c31bec/setup.py#L29C9-L29C35, but it seems that you have version 3 installed.

wiserst, 2024-03-11:
I installed PcbDraw via KiCad 8.0 command line - pip install PcbDraw
Can you please advise me what to do? Thank you.

smkent, 2024-03-24:
https://github.com/yaqwsx/PcbDraw/pull/166 should resolve this.

wiserst, 2024-04-08:
It still doesn't work for me. It will draw a small square in png.
[exam.zip](https://github.com/yaqwsx/PcbDraw/files/14900277/exam.zip)