yaqwsx/PcbDraw #168: pip install package from github not working

lowie2727, 2024-03-31:
when i run the following command

```zsh
pip install git+https://github.com/yaqwsx/PcbDraw
```

i get the following output

```zsh
Collecting git+https://github.com/yaqwsx/PcbDraw
  Cloning https://github.com/yaqwsx/PcbDraw to /tmp/pip-req-build-q3znm4aj
  Running command git clone --filter=blob:none --quiet https://github.com/yaqwsx/PcbDraw /tmp/pip-req-build-q3znm4aj
  Resolved https://github.com/yaqwsx/PcbDraw to commit d2af9d44c998a3ba49c1880aa27bf4479697e5f9
  Running command git submodule update --init --recursive -q
  Installing build dependencies ... done
  Getting requirements to build wheel ... error
  error: subprocess-exited-with-error
  
  × Getting requirements to build wheel did not run successfully.
  │ exit code: 1
  ╰─> [31 lines of output]
      /tmp/pip-req-build-q3znm4aj/versioneer.py:421: SyntaxWarning: invalid escape sequence '\s'
        LONG_VERSION_PY['git'] = '''
      Traceback (most recent call last):
        File "/home/lowie/Downloads/temp/draw/lib64/python3.12/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 353, in <module>
          main()
        File "/home/lowie/Downloads/temp/draw/lib64/python3.12/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 335, in main
          json_out['return_val'] = hook(**hook_input['kwargs'])
                                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "/home/lowie/Downloads/temp/draw/lib64/python3.12/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 118, in get_requires_for_build_wheel
          return hook(config_settings)
                 ^^^^^^^^^^^^^^^^^^^^^
        File "/tmp/pip-build-env-7aymktnk/overlay/lib/python3.12/site-packages/setuptools/build_meta.py", line 325, in get_requires_for_build_wheel
          return self._get_build_requires(config_settings, requirements=['wheel'])
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "/tmp/pip-build-env-7aymktnk/overlay/lib/python3.12/site-packages/setuptools/build_meta.py", line 295, in _get_build_requires
          self.run_setup()
        File "/tmp/pip-build-env-7aymktnk/overlay/lib/python3.12/site-packages/setuptools/build_meta.py", line 487, in run_setup
          super().run_setup(setup_script=setup_script)
        File "/tmp/pip-build-env-7aymktnk/overlay/lib/python3.12/site-packages/setuptools/build_meta.py", line 311, in run_setup
          exec(code, locals())
        File "<string>", line 11, in <module>
        File "/tmp/pip-req-build-q3znm4aj/versioneer.py", line 1480, in get_version
          return get_versions()["version"]
                 ^^^^^^^^^^^^^^
        File "/tmp/pip-req-build-q3znm4aj/versioneer.py", line 1412, in get_versions
          cfg = get_config_from_root(root)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "/tmp/pip-req-build-q3znm4aj/versioneer.py", line 342, in get_config_from_root
          parser = configparser.SafeConfigParser()
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
      AttributeError: module 'configparser' has no attribute 'SafeConfigParser'. Did you mean: 'RawConfigParser'?
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: subprocess-exited-with-error

× Getting requirements to build wheel did not run successfully.
│ exit code: 1
╰─> See above for output.

note: This error originates from a subprocess, and is likely not a problem with pip.
```

Tried fixing it kept getting more errors ;(. I'm on Fedora 39 in a python 3.12.3 venv with the --system-site-packages included.

dreirund, 2024-04-28:
Same here;  
I am on python 3.12.3; python module `configparser` is at version 7.0.0.

mrrb, 2024-04-29:
The latest version of python (3.12) removes some deprecated methods on the ConfigParser class. Check [PR #170](https://github.com/yaqwsx/PcbDraw/pull/170)

yaqwsx, 2024-05-03:
Solved by 1264f28eb0cb96ab7ec44ecb09c5d50eef508ec5, nevertheless, we have some broken code at the moment in the master branch.

closed 2024-05-03 by yaqwsx