yaqwsx/PcbDraw #171: Running PcbDraw on Kicad 8 / W11 gives frozen runpy error

DerRudi63, 2024-05-26:
After installing PcbDraw (and setting the Path accordingly), I can not seem to get PcbDraw even get to produce a help screen. 
This is the command and output I get:

`pcbdraw --help
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "D:\Documents\KiCad\8.0\3rdparty\Python311\Scripts\pcbdraw.exe\__main__.py", line 4, in <module>
  File "D:\Documents\KiCad\8.0\3rdparty\Python311\site-packages\pcbdraw\ui.py", line 15, in <module>
    from .populate import populate
  File "D:\Documents\KiCad\8.0\3rdparty\Python311\site-packages\pcbdraw\populate.py", line 16, in <module>
    from mistune.plugins.table import plugin_table # type: ignore
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
ImportError: cannot import name 'plugin_table' from 'mistune.plugins.table' (D:\Documents\KiCad\8.0\3rdparty\Python311\site-packages\mistune\plugins\table.py)`

Not that familiar with Python, but what can this be? If more info is needed, happy to provide that.
Thanks.

markus-k, 2024-06-16:
Same error in the kikit docker image.

```
Traceback (most recent call last):
  File "/usr/local/bin/pcbdraw", line 5, in <module>
    from pcbdraw.ui import run
  File "/usr/local/lib/python3.10/dist-packages/pcbdraw/ui.py", line 15, in <module>
    from .populate import populate
  File "/usr/local/lib/python3.10/dist-packages/pcbdraw/populate.py", line 16, in <module>
    from mistune.plugins.table import plugin_table # type: ignore
ImportError: cannot import name 'plugin_table' from 'mistune.plugins.table' (/usr/local/lib/python3.10/dist-packages/mistune/plugins/table.py)
```

K0MPLEXWorksTogether, 2024-12-08:
It would be nice if I could get some help on this.

```
Traceback (most recent call last):
  File "/home/k0mplex/Internship/.venv/bin/pcbdraw", line 5, in <module>
    from pcbdraw.ui import run
  File "/home/k0mplex/Internship/.venv/lib/python3.11/site-packages/pcbdraw/ui.py", line 15, in <module>
    from .populate import populate
  File "/home/k0mplex/Internship/.venv/lib/python3.11/site-packages/pcbdraw/populate.py", line 16, in <module>
    from mistune.plugins.table import plugin_table # type: ignore
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
ImportError: cannot import name 'plugin_table' from 'mistune.plugins.table' (/home/k0mplex/Internship/.venv/lib/python3.11/site-packages/mistune/plugins/table.py)
```