yaqwsx/PcbDraw #173: Not working in kicad-nightly-8.99 on Windows 10

wiserst, 2024-07-22:
After failing to get it working with kicad-8.0, I try with kicad-nightly-8.99 and fail again. Can someone help me get this working? I really don't know how to proceed.

I followed the installation instructions and it doesn't work:
https://github.com/yaqwsx/PcbDraw/blob/master/doc/installation.md

If you need financial support, write to me.
Thank you.

---------------------------------------------------------------------------------------------------------------------------------
d:\Renda\Documents\KiCad\8.99>pcbdraw --help
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "d:\Renda\Documents\KiCad\9.0\3rdparty\Python311\Scripts\pcbdraw.exe\__main__.py", line 4, in <module>
  File "d:\Renda\Documents\KiCad\9.0\3rdparty\Python311\site-packages\pcbdraw\ui.py", line 12, in <module>
    from .plot import (PcbPlotter, PlotComponents, PlotPaste, PlotPlaceholders,
  File "d:\Renda\Documents\KiCad\9.0\3rdparty\Python311\site-packages\pcbdraw\plot.py", line 32, in <module>
    from pcbnewTransition import KICAD_VERSION, isV6, isV7, pcbnew # type: ignore
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "d:\Renda\Documents\KiCad\9.0\3rdparty\Python311\site-packages\pcbnewTransition\__init__.py", line 23, in <module>
    from .transition import *
  File "d:\Renda\Documents\KiCad\9.0\3rdparty\Python311\site-packages\pcbnewTransition\transition.py", line 121, in <module>
    pcbnew.PCB_SHAPE = pcbnew.DRAWSEGMENT
                       ^^^^^^^^^^^^^^^^^^
AttributeError: module 'pcbnewTransition.pcbnew' has no attribute 'DRAWSEGMENT'

yaqwsx, 2024-07-22:
KiCAD has unstable API between major releases. Its API changes on a weekly basis. Hence, I don't want to support nightly versions until late November, when the API is getting stable.

wiserst, 2024-07-22:
OK thank you for answer. I'll wait.

wiserst, 2024-12-02:
Hello, Is it already possible to support nightly versions?

wiserst, 2025-02-24:
Hello, is it already possible to support KiCad version 9.0?

MR-DOS, 2025-04-10:
Not working in 9.0.1, either.

SteynGuelen, 2025-06-20:
Also not in 9.0.2. Is this project still being maintained?