yaqwsx/PcbDraw #178: Dependency on Older Python Package Conflicts With KiKit

agatha2, 2024-11-26:
Hi, attempting to install (Windows 10) has a package version conflict with a dependency of kikit.

Python pip will give:
```txt
pcbdraw 1.1.2 requires pcbnewTransition<=0.4,>=0.3, but you have pcbnewtransition 0.4.2 which is incompatible.
```
And if you install anyway, then kikit will similarly complain:
```txt
kikit 1.6.0 requires pcbnewTransition<=0.5,>=0.4.1, but you have pcbnewtransition 0.4.0 which is incompatible.
```
Probably pcbdraw should be updated for later pcbnewTransition. Thanks,

MR-DOS, 2025-04-10:
I confirm this issue. With the latest unstable 9.0.1 KiCAD on Windows 11, installing PcbDraw breaks KiKit due to the dependencies. Also, it's not possible to directly call PcbDraw as it does not register itself into PATH (probably). To fix this, PcbDraw has to be uninstalled and KiKit reinstalled.

set-soft, 2025-05-28:
> This is a huge problem, as its basically made PcbDraw unusable for the past year. I used to have it render images as part of my KiKit processing, but I had to disable all that as a result of this issue.

If you are using a CI/CD environment, or docker, or Linux, and you need a solution now, you can try using [KiBot](https://github.com/INTI-CMNB/KiBot/). It has a built-in PcbDraw that is currently working.